SlimeVR/SlimeVR-Rust #29: Get CI to also check firmware

TheButlah, 2022-08-22:
I had to remove the firmware from the cargo workspace, because it only compiles on RISC-V currently, and has its own `cargo.toml`. I wasn't sure how to do this in a better way, so for now I removed it from the cargo workspace.

I saw that https://github.com/rust-lang/cargo/issues/5220 exists but seems to have no movement.

One solution is for me to have the main.rs have a `cfg(platform)` feature gate. And maybe I could feature gate the dependencies and then activate that feature when on the appropriate architecture?

ImUrX, 2022-12-02:
more simple: make an action for firmware, it will simplify things later when we do want to ``cargo build`` for different archs

TheButlah, 2022-12-02:
> more simple: make an action for firmware

Agreed - this is the right approach. We don't need one single workspace, any code that can't also compile to host architectures can go in firmware workspace.

Long term, maybe we can get the firmware to compile via linux-embedded-linux, but even then, thats platform specific, it won't work on mac or windows.

TheButlah, 2022-12-19:
Closed by #104

closed 2022-12-19 by TheButlah