SlimeVR/SlimeVR-Rust #81: [Overlay] Add logger to rotating file

TheButlah, 2022-11-30:


marcozzxx810, 2023-02-14:
I think I can take this, should be similar to SlimeVR-server. https://github.com/SlimeVR/SlimeVR-Server/issues/558

1. setup two logs (log_last_0.log, log_last_1.log) with 25MB limit
2. rotate the log when the output approximately is equal 25MB size

My plan: 

- [ ] remove [pretty-env-logger](https://github.com/seanmonstar/pretty-env-logger) in the repo since color output is not we wanted any more
- [ ] use [log4rs ](https://github.com/estk/log4rs) for log rotation and log size limiter

TheButlah, 2023-02-14:
@marcozzxx810 I would like to unconditionally log to a file, but conditionally (based on a cli arg) show a terminal popup with the logs there too.

I have never used `log4rs` before but looking at it, it seems quite configurable. It also looks like multiple [appenders](https://docs.rs/log4rs/1.2.0/log4rs/#appenders) can be used at the same time, so maybe it could handle logging to the console and to the file at the same time? Ideally when logging to the console, we would keep the nice pretty colors, but if we need to sacrifice that in order to log to a file at the same time, that is fine.

I appreciate you taking this on! Its very helpful 😊

closed 2023-02-19 by TheButlah