SlimeVR/SlimeVR-Rust #162: [Firmware] ESP32 can't compile because of atomics

ImUrX, 2023-01-26:
It seems [radium is trying to use atomics](https://github.com/SlimeVR/SlimeVR-Rust/actions/runs/4009664483/jobs/6906602412) in ESP32 and its failing to build

TheButlah, 2023-01-27:
```
> cargo tree -i radium
radium v0.7.0
└── bitvec v1.0.1
    └── deku v0.15.1
        └── firmware_protocol v0.0.0 (/home/ryan/P/slimevr/rust/networking/firmware_protocol)
            └── firmware v0.0.0 (/home/ryan/P/slimevr/rust/firmware)
```

TheButlah, 2023-01-27:
So looking into this, radium will try to use atomics if they are natively available, otherwise will use `Cell<T>`.

Radium 1.0.0 uses `#[cfg(target_has_atomic = width)]` to determine if an atomic is supported. This cfg was stabilized in rust 1.60.

Radium 0.7.0 however released before the cfg feature was stabilized. In order to determine if the target supported atomics or not, radium instead used a hack - there was a [`build.rs`](https://github.com/ferrilab/radium/blob/723bed5abd75994ee4b7221b8b12c9f4e77ce408/build.rs#L69-L99) script that would actually hard-code which atomics were supported based on strings gotten from the rustc env. Xtensa is not listed in that build script, so it was probably assuming the wrong atomic support.

If we can have bitvec use radium 1.0.0, either via upstreaming the bump or by patching it ourselves, then assuming that esp-rs rustc does `cfg(target_has_atomic)` correctly on xtensa, we should be good to go.

TheButlah, 2023-01-27:
I opened https://github.com/ferrilab/ferrilab/issues/1

TheButlah, 2023-04-05:
Fixed this and others in #168

closed 2023-04-05 by TheButlah