SlimeVR/SlimeVR-Rust #185: [Overlay] Fix the overlay to align to the first person perspective

TheButlah, 2023-02-12:
We had changes in https://github.com/SlimeVR/SlimeVR-OpenVR-Driver/pull/24 that switched the tracking space from raw to "standing". We need to account for this change in the overlay.

TheButlah, 2023-02-12:
Mentioning @kitlith

kitlith, 2023-02-12:
some of the PRs made to ovr_overlay are necessary for this. (you may as well just assign me for this)

overlay has already been switched to standing, but everything else is now using a variant that ignores playspace movement, which is why changes are still necessary

TheButlah, 2023-02-12:
> some of the PRs made to ovr_overlay are necessary for this.

Working on reviewing them. Could you link the ones that block this issue?

kitlith, 2023-02-12:
https://github.com/TheButlah/ovr_overlay/pull/8 and https://github.com/TheButlah/ovr_overlay/pull/9, though the first one needs to be rebased and worked on a bit now that you merged https://github.com/TheButlah/ovr_overlay/pull/11

I'll get that rebased now.